moveit/geometric_shapes #27: Release into ROS Jade

mikeferguson, 2015-03-15:
Before release, want to:
 * [ ] test and pull other changes from https://github.com/ros-planning/geometric_shapes/pull/25 into indigo-devel

Blocking upstream dependencies:
 * [ ] eigen_stl_containers
 * [ ] octomap
 * [ ] random_numbers
 * [ ] resource_retriever
 * [ ] shape_tools

vrabaud, 2015-03-18:
@isucan, stop getting married and release that ! :)

DLu, 2015-04-14:
I was taking a look at releasing the robot_model stack (of which resource_retriever is a part) and collada_urdf (also in robot_model) depends on geometric_shapes. How should I proceed? Just release resource_retriever?

mikeferguson, 2015-04-19:
Wow -- circular depends between geometric_shapes and robot_model -- I totally did not see that. Checking git-blame on the rosdistro, resource_retriever has apparently been released separately from everything else in the robot_model repo.

The GBP for resource_retriever is here: https://github.com/ros-gbp/resource_retriever-release. Then (Presumably after geometric_shapes was released), the other packages of robot_model got released here: https://github.com/ros-gbp/robot_model-release.

Presumably you should release resource_retriever the same way? @wjwwood It looks like you might have been involved in the robot model releasing for indigo -- any comments on the above?

wjwwood, 2015-04-20:
Looks like I just did a version bump, @isucan did the original release. It looks like he released `resource_retriever` only in the first gbp you linked using an ignore file, then released the rest of `robot_model` using an ignore file to skip `resource_retriever`. The right thing to do here is to move `resource_retriever` to its own repository, but failing that a similar maneuver could be used to release it as is into jade.

isucan, 2015-04-20:
@wjwwood is correct - sorry for being late to this and being lazy about the split. But yes, the repo should ideally be split.

DLu, 2015-04-20:
Is there an easy tool for splitting the repo? (i.e. not just command lines, I can do that)

I presume it would remain in the ros organization as well.

wjwwood, 2015-04-20:
> Is there an easy tool for splitting the repo? (i.e. not just command lines, I can do that)

Not to my knowledge, it's mostly a git (or VCS specific) set of operations. A guide would be good, but unfortunately I think that having a tool that "just does it" would probably be dangerous if used incorrectly.

scpeters, 2015-04-22:
+1, this is blocking `gazebo_ros_pkgs` among other things

wjwwood, 2015-04-22:
Ok, I've moved `resource_retriever` to it's own repository:

https://github.com/ros/resource_retriever.git

and fixed up the indigo release repositories as needed. It was pretty easy since @isucan had already made two release repositories. Here is the pr for the updated release of `resource_retriever` into indigo:

https://github.com/ros/rosdistro/pull/8025

I'll release the Jade versions next.

wjwwood, 2015-04-22:
Jade release: https://github.com/ros/rosdistro/pull/8026

wjwwood, 2015-04-22:
I poked `random_numbers`:

https://github.com/ros-planning/random_numbers/issues/9#issuecomment-94998694

and `octomap`:

https://github.com/OctoMap/octomap/issues/80#issuecomment-94994804

Hopefully we can get the ball rolling on this package and all of `robot_model` tomorrow.

wjwwood, 2015-04-22:
Octomap: https://github.com/ros/rosdistro/pull/8075 :fireworks:

wjwwood, 2015-04-22:
@mikeferguson I've poked @davetcoleman about either releasing random_numbers or giving me the go ahead to do it myself. After that the only thing blocking a Jade release is your mention of #25. Is it important to get that in the first release for Jade?

mikeferguson, 2015-04-22:
@wjwwood if we release from the indigo-devel branch, it is not important -- basically I didn't want to fork until after that PR was merged.

mikeferguson, 2015-04-22:
I'll watch for random_numbers on the rosdistro and then release this as soon as that comes through

scpeters, 2015-04-22:
I was getting a build failure on utopic without the following:
~~~
diff --git a/package.xml b/package.xml
index d9b2c0b..ee8191d 100644
--- a/package.xml
+++ b/package.xml
@@ -36,5 +36,6 @@
   <run_depend>random_numbers</run_depend>
   <run_depend>eigen_stl_containers</run_depend>
   <run_depend>resource_retriever</run_depend>
+  <run_depend>visualization_msgs</run_depend>
 
 </package>
~~~
This is addressed by #32. Will #32 be merged before the release?

scpeters, 2015-04-22:
I just made #33 to fix the build if #32 won't be merged before release.

wjwwood, 2015-04-22:
Ok, `random_numbers` is out: https://github.com/ros/rosdistro/pull/8079 :package: :shipit:

mikeferguson, 2015-04-22:
https://github.com/ros/rosdistro/pull/8080

closed 2015-04-22 by mikeferguson

davetcoleman, 2015-04-22:
Great, thanks guys!

on the note of releasing MoveIt! - I'm buried in preparing for the Amazon Picking Challenges so I'm not sure how much help I can provide