moveit/geometric_shapes #42: Migrating from fcl to moveit-geometric shapes

dblanm, 2016-06-22:
Hello everyone,

I was using before the fcl libraries, but after founding several problems I would like to use this library.
The code I had before was:
_typedef std::shared_ptr <fcl::CollisionGeometry> CollisionGeometryPtr_t;
// Define a Ellipsoid with the dimensions of the joint+link
  //This geometry may be modified depending on the robot
  CollisionGeometryPtr_t boxGeometry (new fcl::Box (16.,16.,8.));_

The one for using this library would be:
_shapes::Shape *shape = new shapes::Box(16.0, 16.0, 8.0);_
Without any collisiongeometry defined. Now how can I define a collisionGeometry to check if two boxes collide?
Thank you in advance

davetcoleman, 2016-10-17:
Did you find a solution to this? MoveIt!'s planning scene is how most of us use this package and FCL to check collisions:

https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h

dblanm, 2016-10-17:
I started to use the whole FCL library, so I stopped using MoveIt and the library reshape of FCL functions.

Sorry!

closed 2016-10-17 by davetcoleman