moveit/geometric_shapes #48: Release into kinetic

jonbinney, 2016-07-30:
Currently version `0.4.4` is released into kinetic, but a problem with the shared_ptr types has since been fixed.

130s, 2016-07-31:
Done https://github.com/ros/rosdistro/pull/12246

Note that I bumped the minor version and it's now `0.5.0` for Kinetic, because 1) now we're using `kinetic-devel` branch separate from `indigo-devel` 2) `kinetic-devel` seems to diverge (C++11, FCL, eigen etc.)

130s, 2016-08-02:
Sorry, release request is reverted.

From https://github.com/ros/rosdistro/pull/12246#issuecomment-236726117

> This appears to use c++11 in it's headers but doesn't export it for downstream. Causing failures:

> ```
01:11:58 In file included from /tmp/binarydeb/ros-kinetic-collada-urdf-1.12.3/src/collada_urdf.cpp:52:0:
01:11:58 /opt/ros/kinetic/include/resource_retriever/retriever.h:81:38: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
01:11:58    Retriever(const Retriever & ret) = delete;
01:11:58                                       ^
01:11:58 In file included from /tmp/binarydeb/ros-kinetic-collada-urdf-1.12.3/src/collada_urdf.cpp:82:0:
01:11:58 /opt/ros/kinetic/include/geometric_shapes/shapes.h:254:21: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
01:11:58    OcTree(const std::shared_ptr<const octomap::OcTree> &t);
01:11:58                      ^
01:11:58 /opt/ros/kinetic/include/geometric_shapes/shapes.h:254:31: error: expected ‘,’ or ‘...’ before ‘<’ token
01:11:58    OcTree(const std::shared_ptr<const octomap::OcTree> &t);
01:11:58                                ^
01:11:58 /opt/ros/kinetic/include/geometric_shapes/shapes.h:264:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
01:11:58    std::shared_ptr<const octomap::OcTree> octree;
01:11:58         ^
01:11:58 CMakeFiles/collada_urdf.dir/build.make:65: recipe for target 
> ```
>
> http://build.ros.org/view/Kbin_uX32/job/Kbin_uX64__collada_urdf__ubuntu_xenial_amd64__binary/16/console

130s, 2016-08-02:
For some reasons this time I slacked off running a pre-release test. Once #49 gets resolved, I'll make a new release with pre-release test on.

130s, 2016-09-02:
Looks like 0.5.1 was released several days ago https://github.com/ros/rosdistro/pull/12471 and it seems building fine ([example](http://build.ros.org/view/Kbin_uX32/job/Kbin_uX64__collada_urdf__ubuntu_xenial_amd64__binary/19/)).

closed 2016-09-02 by 130s