moveit/geometric_shapes #61: Release 0.5.2

130s, 2016-10-20:
Suggested at https://github.com/ros-planning/moveit/issues/18#issuecomment-254908832

- [x] changelog, tagging
- [ ] prerelease test (wily, xenial). Running prerelease test is very preferred (see #48 where prerelease test could have avoided false release)
- [ ] bloom 

I'm stuck at running prerelease test by https://github.com/ros-infrastructure/ros_buildfarm/issues/347, which *may* be only happening on my Ubuntu. If someone could run the following command, and if that runs fine then share the result, maybe the last 20 lines of it.

```
sudo apt-get install python-ros-buildfarm
dpkg -p python-ros-buildfarm |grep Version
mkdir /tmp/prerelease1 && cd /tmp/prerelease1
generate_prerelease_script.py https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml kinetic default ubuntu xenial amd64 geometric_shapes  --level 0   --output-dir ./
./prerelease.sh
```

On another terminal,
```
generate_prerelease_script.py https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml kinetic default ubuntu wily amd64 geometric_shapes  --level 0   --output-dir ./
./prerelease.sh
```

130s, 2016-10-20:
To run a prerelease test locally you need to [set up your machine](http://wiki.ros.org/bloom/Tutorials/PrereleaseTest).

davetcoleman, 2016-10-20:
I got several errors also:

```
Setting up python-ros-buildfarm (1.2.0-1) ...
[ros-monster] ~$ dpkg -p python-ros-buildfarm |grep Version
dpkg-query: package 'python-ros-buildfarm' is not available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
[ros-monster] ~$ mkdir /tmp/prerelease1 && cd /tmp/prerelease1
[ros-monster] prerelease1$ generate_prerelease_script.py https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml kinetic default ubuntu xenial amd64 geometric_shapes  --level 0   --output-dir ./
Fetching buildfarm configuration...
Fetching rosdistro cache...
Evaluating job templates...
IOError processing template 'devel/devel_job.xml.em'
Traceback (most recent call last):
  File "/usr/bin/generate_prerelease_script.py", line 299, in <module>
    sys.exit(main())
  File "/usr/bin/generate_prerelease_script.py", line 193, in main
    build_targets=release_targets_combined)
  File "/usr/lib/python2.7/dist-packages/ros_buildfarm/devel_job.py", line 305, in configure_devel_job
    is_disabled=is_disabled)
  File "/usr/lib/python2.7/dist-packages/ros_buildfarm/devel_job.py", line 392, in _get_devel_job_config
    job_config = expand_template(template_name, job_data)
  File "/usr/lib/python2.7/dist-packages/ros_buildfarm/templates/__init__.py", line 92, in expand_template
    data['wrapper_scripts'] = get_wrapper_scripts()
  File "/usr/lib/python2.7/dist-packages/ros_buildfarm/templates/__init__.py", line 168, in get_wrapper_scripts
    with open(abs_file_path, 'r') as h:
IOError: [Errno 2] No such file or directory: '/usr/lib/python2.7/dist-packages/ros_buildfarm/wrapper/apt.py'
```

130s, 2016-10-20:
One thought is to make a judgment call to skip prerelease test only this time and see if ROS buildfarm builds it without waiting for fixing the prerelease test issue (so that we can move forward with moveit Kinetic release https://github.com/ros-planning/moveit/issues/18). If there's an issue on buildfarm we can quick-fix and make a release again.

I'll make a release now.

130s, 2016-10-20:
Release requested for Kinetic https://github.com/ros/rosdistro/pull/12984. As said above I skipped prerelease test this time due to the tight timeframe for MoveIt! Kinetic release.

closed 2016-10-20 by 130s

davetcoleman, 2016-10-20:
Thanks @130s !

130s, 2016-10-20:
Just fyi I ran a prerelease test now that https://github.com/ros-infrastructure/ros_buildfarm/issues/347 is resolved and it seems fine for Wily and Xenial.

```
$ dpkg -p python-ros-buildfarm |grep Ver
Version: 1.2.1-1

$ generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml   kinetic default ubuntu xenial amd64   geometric_shapes   --level 0   --output-dir ./
$  ./prerelease.sh 
:
# END SUBSECTION
+ echo # END SECTION
# END SECTION

Build step 8
+ [  = false ]

No test results in overlay workspace

$ generate_prerelease_script.py   https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml   kinetic default ubuntu wily amd64   geometric_shapes   --level 0   --output-dir ./
$  ./prerelease.sh
:
# END SUBSECTION
+ echo # END SECTION
# END SECTION

Build step 8
+ [  = false ]

No test results in overlay workspace
```

Buildfarm also seems happy http://repositories.ros.org/status_page/ros_kinetic_default.html?q=geometric_shapes