moveit/geometric_shapes #65: Lunar release

mikaelarguedas, 2017-04-06:
Hi @isucan, @davetcoleman and geometric_shapes maintainers,

As you may know the next ROS release Lunar Loggerhead is around the corner :tada: 
Is it possible to roll out the current release of `geometric_shapes` on ROS Lunar? Being a low level package it will [allow many maintainers](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22geometric_shapes%22&s=6&r=1) to release their own packages that depend directly or indirectly on geometric_shapes.

Thanks!

davetcoleman, 2017-04-10:
@130s typically takes care of releases for the moveit-related packages

I'll also remove @isucan from the package.xml, as he is no longer involved

mikaelarguedas, 2017-04-10:
Thanks @davetcoleman for the update.
@130s please comment here if you don't have any spare cycle to release this today. Given the impact of this on every other package, I'll bloom the current kinetic version myself if it hasn't been released by the end of the day.
Cheers

davetcoleman, 2017-04-10:
I'll release it now...

mikaelarguedas, 2017-04-10:
thanks @davetcoleman !

davetcoleman, 2017-04-10:
was not able to get past rosdep, feel free to take over or i can continue in a few hours.

https://github.com/ros-infrastructure/rosdep/issues/508

wjwwood, 2017-04-10:
Why do you need local rules to run? You were just waiting on https://github.com/ros/rosdistro/pull/14498 to get merged?

mikaelarguedas, 2017-04-10:
Thanks @davetcoleman for spending time on it today! I followed up and opened https://github.com/ros/rosdistro/issues/14501

mikaelarguedas, 2017-04-11:
Compiled without pb on all platforms :tada:

closed 2017-04-11 by mikaelarguedas

davetcoleman, 2017-04-11:
@wjwwood even after https://github.com/ros/rosdistro/pull/14498 was merged I was still getting errors about missing dependencies, so my plan was to fix them all locally before creating the next PR. But it seems it just didn't catch the latest PR, because I think @mikaelarguedas was able to release without further rosdep rules.