moveit/geometric_shapes #69: Can't find Eigen3 when compiling Kinetic from sources

VictorLamoine, 2017-06-10:
I'm following the [build from source](wiki.ros.org/kinetic/Installation/Source) instructions, I'm using a Raspberry Pi 3 with Raspbian.

```bash
==> cmake /home/victor/libraries/ros_catkin_ws/src/geometric_shapes -DCATKIN_DEVEL_PREFIX=/home/victor/libraries/ros_catkin_ws/devel_isolated/geometric_shapes -DCMAKE_INSTALL_PREFIX=/home/victor/libraries/ros_catkin_ws/install_isolated -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/home/victor/libraries/ros_catkin_ws/build_isolated/geometric_shapes'
-- Checking for module 'assimp'
--   Found assimp, version 3.0.1264
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   system
--   filesystem
CMake Error at CMakeLists.txt:24 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

```bash
$ sudo find / -iname "FindEigen3.cmake"
/home/victor/libraries/ros_catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/cmake/FindEigen3.cmake
/home/victor/libraries/ros_catkin_ws/src/orocos_kinematics_dynamics/orocos_kdl/config/FindEigen3.cmake
/usr/share/cmake-3.0/Modules/FindEigen3.cmake
/usr/share/cmake-2.8/Modules/FindEigen3.cmake
```

```bash
$ cmake --version
cmake version 3.6.2

```

```bash
 $ dpkg -l | grep eigen
ii  libeigen3-dev  3.2.2-3  all  lightweight C++ template library for linear algebra
```

I solved the issue with:
```bash
sudo ln -s /usr/share/cmake-3.0/Modules/FindEigen3.cmake /usr/share/cmake-3.6/Modules/FindEigen3.cmake
```

Is it the fault of Eigen package not installing it's FindEigen3.cmake to the right path?

de-vri-es, 2017-06-10:
It shouldn't install `FindEigen3.cmake` at all, it should install `Eigen3Config.cmake`. But yeah, it sounds like a Raspian packaging problem. Your best bet would be opening a bug report with them.

VictorLamoine, 2017-06-10:
Right, CMake provides the `FindEigen3.cmake` macro!
Closing as this is not related to `geometric_shapes`

closed 2017-06-10 by VictorLamoine