moveit/geometric_shapes #78: Release into Melodic

clalancette, 2018-04-05:
All of the dependencies for geometric_shapes are available in Melodic, so it would be great to get this released there.  Thanks in advance!

rhaschke, 2018-04-06:
@clalancette I would do so. But I don't have permissions for the release repo.

rhaschke, 2018-04-06:
TODOs for release(s)

- [ ] permission for release repo https://github.com/ros-gbp/geometric_shapes-release.git
- [ ] merge #81 to fix #64 
- [ ] release 0.5.4 into kinetic
- [ ] new melodic-devel branch
- [ ] merge #77 into melodic-devel
- [ ] release 0.6.0 into melodic (API/ABI changes)

rhaschke, 2018-04-06:
@bmagyar I prepared the release 0.5.4, creating the `changelog` and tag `0.5.4`. If you could `bloom` into Kinetic, that would be great, thanks.
I also suggest to release this version (`0.5.4`) into Melodic for now, so that releasing for ROS Melodic can continue. If #77 is merged, we should release the new version `0.6.0`.

bmagyar, 2018-04-06:
Thanks! I pushed 0.5.4 to Kinetic, already merged. Ran pre-release tests with dependants for Lunar (took a while!), released there too. Added 0.5.4 to Melodic as well. 

Kinetic and Lunar use `kinetic-devel` while the Melodic release already uses `melodic-devel` for devel branch. We should be good to move on with merges and newer versions soon.

BryceStevenWilley, 2019-05-01:
Closing this, as we've released.

closed 2019-05-01 by BryceStevenWilley