moveit/geometric_shapes #83: wstool/rosdep fetches incompatible versions of geometric_shapes and libconsole-bridge

jeholliday, 2018-04-19:
I ran into compatibility issues when trying to build this package with ROS from source on armhf Debian jessie following [this guide](http://wiki.ros.org/kinetic/Installation/Debian). The problem came from rosdep fetching the latest version it could find of libconsole-bridge which is only [0.2.5-2](https://packages.debian.org/jessie/libdevel/libconsole-bridge-dev). #72 removed support for this old version of libconsole-bridge. In order to get the package to compile, I removed the latest version of geometric_shapes which was automatically fetched by wstool and replaced it with version [0.5.3](https://github.com/ros-planning/geometric_shapes/releases/tag/0.5.3), from before #72.

Is there a way which you can define that geometric_shapes depends on a newer version of console-bridge to prevent people running into this issue in the future?

rhaschke, 2018-04-20:
Obviously Jessie is way too old to compile a decent MoveIt version without issues.
I'm not sure what your are asking for. Yes, one could enforce geometric_shapes to require libconsole-bridge 0.3. However, that's not the way you finally solved your issue...

jeholliday, 2018-04-20:
I'm not familiar with how wstool and rosdep work, but it seems like they need to know that it should only fetch an older version of geometric_shapes when it can't find the latest version of console-bridge. I don't know how you implement this, but I wanted to let you know.

BryceStevenWilley, 2019-05-01:
Not sure if wstool/rosdep work that way. wstool only provides for a single specific version in a rosinstall file (either a specific commit or a branch), and limiting `geometric_shapes` to a commit pre #72 is a no-go.  rosdep can specify `version_gt`, of which `libconsole-bridge-dev` could be `> 0.4.0`, but as @rhaschke said, that wouldn't fix your problem, and rosdep wouldn't necessarily have [enforced the version dependency](https://github.com/ros-infrastructure/rosdep/issues/325) correctly either. The most that would happen is that [catkin would print a warning](https://answers.ros.org/question/280048/package-manifest-2-depend-attribute-version-does-not-work/), which I think code that fails to compile is probably a better indicator of mismatching versions than a warning. The best way to resolve issues like this is how you did it: by manually checking out the version of geometric_shapes that worked. 

I'm going to close this as won't fix, feel free to reopen if you disagree.

closed 2019-05-01 by BryceStevenWilley