moveit/geometric_shapes #87: Shape Derived Classes Lack Co-Variant Return Type for Clone()

Jmeyer1292, 2018-09-01:
C++ has covariant return types. They are useful for things. They are especially useful for implementing `clone()` functions polymorphically. See [this page](http://www.lwithers.me.uk/articles/covariant.html).

So for instance:
```c++
class Mesh : public Shape
{
//...
virtual Shape* clone() const;
// ...
};
```
should be
```c++
class Mesh : public Shape
{
//...
virtual Mesh* clone() const;
// ...
};
```
I was attempting to load many, large models using this library. I wanted to cache the `shapes::Mesh*` pointers using their URI name. Unfortunately even though I know I have a Mesh, I can't clone it to a mesh because the function returns the base class. This currently requires an unneeded dynamic_cast to fix.

v4hn, 2018-09-01:
Sounds good. Could you provide a patch please?

rhaschke, 2019-08-08:
Fixed in #102.

closed 2019-08-08 by rhaschke