moveit/geometric_shapes #131: Release geometric_shapes into ROS Noetic?

sloretz, 2020-04-13:
Would you mind releasing `geometric_shapes` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 41 repos need it.

simonschmeisser, 2020-04-23:
probably also something for your weekend @rhaschke ?

rhaschke, 2020-04-24:
Let's see how much time my weekend actually has...
It would be great of course to have all the pending PRs merged before a release... @peci1, @v4hn what is missing to this end?

v4hn, 2020-04-24:
On Thu, Apr 23, 2020 at 08:45:22PM -0700, Robert Haschke wrote:
> Let's see how much time my weekend actually has...
> It would be great of course to have all the pending PRs merged before a release... @peci1, @v4hn what is missing to this end?

Someone to review and test them.
In my opinion, you might release it already to enable dependency resolution though.
We should have a followup release soon though.

peci1, 2020-04-24:
Yeah, I think all my PRs are ready to be merged and reviews are the only missing thing. I'll be available today and on Saturday. On Sunday I probably won't be able to quickly respond...

v4hn, 2020-04-24:
I spend half the day reviewing all open requests here and peci just addressed my few comments.

There are 6 open and approved pull-requests lying around waiting for a second approval to merge.

peci1, 2020-04-26:
Would it be possible to leave at least a week or so since noetic-devel is created to the actual API/ABI freeze. There are things that'd benefit from breaking API/ABI, but as there is no noetic-devel yet, it's impossible to send PRs...

rhaschke, 2020-04-26:
No problem to postpone until next week. geometric_shapes is mainly [blocking the MoveIt release](http://repositories.ros.org/status_page/blocked_releases_noetic.html?q=geometric_shapes&s=6&r=1), which is not going to be released into Noetic within the next few weeks.

sloretz, 2020-05-12:
Friendly ping, anything `geometric_shapes` needs before release?

rhaschke, 2020-05-12:
@v4hn, do we want to merge the open PR(s) before a Noetic release?

wxmerkt, 2020-05-22:
Friendly ping, are there other PRs you'd like to merge before releasing? I am happy to help out with testing or in another capacity :-)

peci1, 2020-05-22:
I think #144 should definitely be merged before release. It was requested in https://github.com/ros-planning/moveit/pull/1964 and the changes in moveit will depend on it.

rhaschke, 2020-05-25:
Release PR: https://github.com/ros/rosdistro/pull/25051

closed 2020-05-25 by rhaschke