moveit/geometric_shapes #152: Drop -devel from release branch names

tylerjw, 2020-06-24:
As these branches are used for releases and not current devleopment, could we just use the ros version as the branch name?

v4hn, 2020-06-24:
> As these branches are used for releases and not current devleopment, could we just use the ros version as the branch name?

The `-devel` suffix is rather a ROS standard to indicate there might be (and usually are) more commits in there than in the last version released into the distribution.

It's just a convention, true.
But I consider it useful to refer to when talking to users who tell you "I used the kinetic version". Well, which one? The release or the upstream repo branch?

tylerjw, 2020-06-24:
There are only unreleased features in these branches when we are preparing for a release though right?

v4hn, 2020-06-25:
> There are only unreleased features in these branches when we are preparing for a release though right?

That description includes almost always - except for the time span after the release and before someone requests a backport (or someone gets around to look through the backport commit log).
Of course over time, less users of a distribution are interested in backports, but it still happens rather regularly.

closed 2020-06-25 by tylerjw