moveit/geometric_shapes #166: foxy build error linking boost libbrary

zuxinrui, 2021-03-03:
**Platform:** ubuntu 20.04 foxy
When build the package from source using ros2 branch, something wrong with linking the boost library:

--- stderr: geometric_shapes
/usr/bin/ld: CMakeFiles/test_loaded_meshes.dir/test_loaded_meshes.cpp.o: in function `CompareMeshVsPrimitive::SetUp()':
test_loaded_meshes.cpp:(.text._ZN22CompareMeshVsPrimitive5SetUpEv[_ZN22CompareMeshVsPrimitive5SetUpEv]+0x13f): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_loaded_meshes.dir/build.make:158: test/test_loaded_meshes] Error 1
make[1]: *** [CMakeFiles/Makefile2:422: test/CMakeFiles/test_loaded_meshes.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/test_point_inclusion.dir/test_point_inclusion.cpp.o: in function `MeshPointContainment_Pr2Forearm_Test::TestBody()':
test_point_inclusion.cpp:(.text+0x710b): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/test_point_inclusion.dir/test_point_inclusion.cpp.o: in function `MeshPointContainment_Basic_Test::TestBody()':
test_point_inclusion.cpp:(.text+0x17216): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_point_inclusion.dir/build.make:158: test/test_point_inclusion] Error 1
make[1]: *** [CMakeFiles/Makefile2:277: test/CMakeFiles/test_point_inclusion.dir/all] Error 2
/usr/bin/ld: CMakeFiles/test_body_operations.dir/test_body_operations.cpp.o: in function `Bodies_ConstructShapeFromBodyMesh_Test::TestBody()':
test_body_operations.cpp:(.text+0x2346): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/test_body_operations.dir/test_body_operations.cpp.o: in function `Bodies_ConstructMarkerFromBodyMesh_Test::TestBody()':
test_body_operations.cpp:(.text+0x341a): undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_body_operations.dir/build.make:158: test/test_body_operations] Error 1
make[1]: *** [CMakeFiles/Makefile2:480: test/CMakeFiles/test_body_operations.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

I'm not sure if this can be fixed with adjusting the Cmakelist.txt

pvanlaar, 2021-06-10:
I tried to reproduce the issue, but I could not. I started from a clean fox-ros-base installation on Ubuntu 20.04 and cloned this repo. After that I used rosdep to install the dependencies for this package. One of the dependencies that got installed was Boost (installed as version 17, i.e. libboost1.71-dev).

Could you be missing libboost-dev on your system?

### Commands you could use:
Install Boost using apt-get
~~~
sudo apt-get update -qq
sudo apt-get install libboost1.71-dev
~~~

Using rosdep to install all dependencies:
~~~bash
sudo rosdep init
rosdep update # no sudo

## from workspace root
rosdep install --from-paths src -y -i
~~~

JafarAbdi, 2021-11-01:
Should be fixed with https://github.com/ros-planning/geometric_shapes/pull/205

closed 2021-11-01 by JafarAbdi