moveit/geometric_shapes #185: --- stderr: geometric_shapes/src/bodies.cpp:44: warning: redundant redeclaration(s)

newcanopies, 2021-04-30:
Hi, should this error be logged when building `geometric_shapes `  and does it affect package functionality?

```
--- stderr: geometric_shapes
In file included from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/libqhull_r.h:1057:9: warning: redundant redeclaration of ‘void qh_printsummary(qhT*, FILE*)’ in same scope [-Wredundant-decls]
 1057 | void    qh_printsummary(qhT *qh, FILE *fp);
      |         ^~~~~~~~~~~~~~~
/usr/include/libqhull_r/libqhull_r.h:1024:9: note: previous declaration of ‘void qh_printsummary(qhT*, FILE*)’
 1024 | void    qh_printsummary(qhT *qh, FILE *fp);
      |         ^~~~~~~~~~~~~~~
/usr/include/libqhull_r/libqhull_r.h:1099:6: warning: redundant redeclaration of ‘void qh_meminit(qhT*, FILE*)’ in same scope [-Wredundant-decls]
 1099 | void qh_meminit(qhT *qh, FILE *ferr);
      |      ^~~~~~~~~~
In file included from /usr/include/libqhull_r/libqhull_r.h:31,
                 from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/mem_r.h:218:6: note: previous declaration of ‘void qh_meminit(qhT*, FILE*)’
  218 | void qh_meminit(qhT *qh, FILE *ferr);
      |      ^~~~~~~~~~
In file included from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/libqhull_r.h:1100:6: warning: redundant redeclaration of ‘void qh_memfreeshort(qhT*, int*, int*)’ in same scope [-Wredundant-decls]
 1100 | void qh_memfreeshort(qhT *qh, int *curlong, int *totlong);
      |      ^~~~~~~~~~~~~~~
In file included from /usr/include/libqhull_r/libqhull_r.h:31,
                 from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/mem_r.h:217:6: note: previous declaration of ‘void qh_memfreeshort(qhT*, int*, int*)’
  217 | void qh_memfreeshort(qhT *qh, int *curlong, int *totlong);
      |      ^~~~~~~~~~~~~~~
In file included from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/libqhull_r.h:1123:9: warning: redundant redeclaration of ‘void qh_collectstatistics(qhT*)’ in same scope [-Wredundant-decls]
 1123 | void    qh_collectstatistics(qhT *qh);
      |         ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/libqhull_r/libqhull_r.h:121,
                 from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/stat_r.h:515:9: note: previous declaration of ‘void qh_collectstatistics(qhT*)’
  515 | void    qh_collectstatistics(qhT *qh);
      |         ^~~~~~~~~~~~~~~~~~~~
In file included from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/libqhull_r.h:1124:9: warning: redundant redeclaration of ‘void qh_printallstatistics(qhT*, FILE*, const char*)’ in same scope [-Wredundant-decls]
 1124 | void    qh_printallstatistics(qhT *qh, FILE *fp, const char *string);
      |         ^~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/libqhull_r/libqhull_r.h:121,
                 from /home/user/ws_ROS2/src/moveit2/geometric_shapes/src/bodies.cpp:44:
/usr/include/libqhull_r/stat_r.h:519:9: note: previous declaration of ‘void qh_printallstatistics(qhT*, FILE*, const char*)’
  519 | void    qh_printallstatistics(qhT *qh, FILE *fp, const char *string);
      |         ^~~~~~~~~~~~~~~~~~~~~
---
```

## env
  - Ubuntu 20.04
  - ROS2 Foxy
  - binaries
  - server glx vendor string: SGI
  - sudo apt update = up to date

tylerjw, 2021-04-30:
Those warnings are from the `qhull` library headers.  That is not part of this project and we do see those warnings every time we build.  If you have any ideas for how to to fix those warnings we'd appreciate it.

rhaschke, 2021-04-30:
`qhull` headers (as all other external headers) should be included via `-isystem` instead of using `-I`. This should silence any warnings outside the code base.

rhaschke, 2021-05-19:
Fixed via #186

closed 2021-05-19 by rhaschke