moveit/geometric_shapes #199: building geometric_shapes results in error /visualization_msgs/msg/detail/markertraits.hpp

toomuchcookies, 2021-06-25:
Following the [moveit getting started tutorial](http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html) but substituting 'galactic' for 'foxy' i got this error message while building:

https://gist.github.com/toomuchcookies/22dd60fde7e6e5691956c16efd4f2f3e

@JafarAbdi , i hope this helps.

vatanaksoytezer, 2021-06-27:
Hi @toomuchcookies thanks for reporting the issue! I tried this with a fresh galactic docker that contains `ros-galactic-desktop`. Following the instructions on getting started compiles moveit2 (including geometric_shapes) without any problems. Did you also perform the actions in installing ros2 and colcon part of getting started http://moveit2_tutorials.picknik.ai/doc/getting_started/getting_started.html#install-ros2-and-colcon? I also advise you to use moveit2_galactic.repos instead of moveit2.repos since there are some unreleased upstream packages that we use in Galactic. 

I also found there are some compile issues with `rviz_visual_tools` on galactic and will work on a fix today, currently that prevents compile of moveit2_tutorials on galactic.

On a related note everything should work on Foxy smoothly.

vatanaksoytezer, 2021-06-27:
https://github.com/PickNikRobotics/rviz_visual_tools/pull/190 should fix compile issues and enable build of moveit2_tutorials on Galactic

toomuchcookies, 2021-06-30:
So i first tried to start anew and followed the tutorial using foxy. Unfortunately now there is a new error:
https://gist.github.com/toomuchcookies/181501d4fbaf818a3f341afaa681736b

I will try to build for galactic and see if the main issue is already solved.

toomuchcookies, 2021-06-30:
oh, just realized these are "only" warnings.. will continue with foxy..

vatanaksoytezer, 2021-06-30:
> oh, just realized these are "only" warnings.. will continue with foxy.

We will spend some time to fix warnings after the new Galactic / Rolling release today, but you can safely ignore them for now.

I am going to close this issue since the problem should not occur right now. Please open a new issue if you face any difficulties and thanks again for reporting!

closed 2021-06-30 by vatanaksoytezer