moveit/geometric_shapes #222: colcon build error ros2 humble

AltinayGrass, 2022-05-25:
Hello all,

There is a source build error on this package like below;

--- stderr: geometric_shapes
CMake Error at CMakeLists.txt:44 (find_package):
  By not providing "Findeigen_stl_containers.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "eigen_stl_containers", but CMake did not find one.

  Could not find a package configuration file provided by
  "eigen_stl_containers" with any of the following names:

    eigen_stl_containersConfig.cmake
    eigen_stl_containers-config.cmake

  Add the installation prefix of "eigen_stl_containers" to CMAKE_PREFIX_PATH
  or set "eigen_stl_containers_DIR" to a directory containing one of the
  above files.  If "eigen_stl_containers" provides a separate development
  package or SDK, be sure it has been installed.

---
Failed   <<< geometric_shapes [0.85s, exited with code 1]

vatanaksoytezer, 2022-05-25:
Hello @AltinayGrass thank you for reporting an issue. Any chance you missed the rosdep step as in: rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y?

AltinayGrass, 2022-05-25:
Merhaba @vatanaksoytezer,
No I am sure we did not miss. But we tried again and the output is:

~/ws_moveit2$ rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
#All required rosdeps installed successfully

vatanaksoytezer, 2022-05-25:
Assuming you are sourcing humble correctly, `source /opt/ros/humble/setup.bash`. Can you try installing eigen-stl-containers via `sudo apt-get install ros-humble-eigen-stl-containers`?

AltinayGrass, 2022-05-25:
we built from source so that our sourcing `source ~/ros2_humble/install/local_setup.bash`

ros-humble-eigen3-cmake and ros-humble-eigen-stl... packages are already installed.

AltinayGrass, 2022-05-25:
One last word;

We have 0 size file at `/home/as/ros2_humble/install/eigen3_cmake_module/share/colcon-core/packages`  `eigen3_cmake_module`

we build a single package with following command. 

`colcon build --packages-select eigen3_cmake_module --allow-overriding  eigen3_cmake_module `

Output: 

Starting >>> eigen3_cmake_module
Finished <<< eigen3_cmake_module [0.33s]
                  
Summary: 1 package finished [1.10s] 

may be this error is related with ros2 humble source build

JafarAbdi, 2022-05-26:
> we built from source so that our sourcing source ~/ros2_humble/install/local_setup.bash

What's the value for the following env-variables before you source your workspace? Please ensure they don't reference any other workspace in your system.

`CMAKE_PREFIX_PATH AMENT_PREFIX_PATH AMENT_CURRENT_PREFIX COLCON_PREFIX_PATH`

AltinayGrass, 2022-05-26:
> What's the value for the following env-variables before you source your workspace? Please ensure they don't reference any other workspace in your system.
>
> `CMAKE_PREFIX_PATH AMENT_PREFIX_PATH [...]

all off them are Null.

Only `AMENT_PREFIX_PATH=/opt/ros/humble`  has value after sourcing.

I tried to install ros2 humble from binary also. Demo_nodes are working fine.

But we are getting some minor warnings like below while in building process ; after a while terminal window is closing itself so I can't see final errors.

--- stderr: moveit_configs_utils
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---

closed 2025-06-06 by rhaschke