moveit/geometric_shapes #226: Could not load resource mesh rendering error

quarkytale, 2022-11-10:
Hi, need some help figuring out the cause of this issue. I have my moveit installation from source in Humble and Ubuntu 22.04. Trying to use the setup_assistant for planning assets of an arm.
```
[moveit_setup_assistant-1] Error:   Error retrieving file [/home/user/demos_ws/src/simulation/models/canadarm/meshes/joint_v3_0.dae]: 
[moveit_setup_assistant-1]          at line 277 in /home/user/moveit2/src/geometric_shapes/src/mesh_operations.cpp
[moveit_setup_assistant-1] [rviz_rendering:error] Could not load resource [/home/user/demos_ws/src/simulation/models/canadarm/meshes/joint_v3_0.dae]: Unable to open file "/home/user/demos_ws/src/simulation/models/canadarm/meshes/joint_v3_0.dae"., at /home/user/moveit2/src/rviz/rviz_rendering/src/rviz_rendering/mesh_loader.cpp:131
```

rhaschke, 2022-11-10:
I think the error message is very clear about the issue:
Unable to open file "/home/user/demos_ws/src/simulation/models/canadarm/meshes/joint_v3_0.dae

Does that file exist? Is it readable for you?

quarkytale, 2022-11-10:
Yes I tried viewing it directly using `g3dviewer`, here's how it looks.
[image]

`assimp info <path to file>` gives:
```
Importing file ...                   OK 
   import took approx. 0.00448 seconds

Memory consumption: 93800 B
Nodes:              2
Maximum depth       2
Meshes:             3
Animations:         0
Textures (embed.):  0
Materials:          1
Cameras:            0
Lights:             0
Vertices:           1135
Faces:              646
Bones:              0
Animation Channels: 0
Primitive Types:    pointslinestriangles
Average faces/mesh  215
Average verts/mesh  378
Minimum point      (-0.175000 -0.000000 -0.140773)
Maximum point      (0.175000 0.800090 0.140773)
Center point       (0.000000 0.400045 0.000000)

Meshes:  (name) [vertices / bones / faces | primitive_types]
    0 (Cylinder-mesh): [1 / 0 / 1 | point]
    1 (Cylinder-mesh): [6 / 0 / 3 | line]
    2 (Cylinder-mesh): [1128 / 0 / 642 | triangle]

Named Materials:
    'Material'

Texture Refs:
    'joint_v3_0_tex.png'

Node hierarchy:
Scene
└╴Cylinder (mesh 0, 1, 2)
```

rhaschke, 2023-03-31:
Upstream error. Closing here.

closed 2023-03-31 by rhaschke