moveit/geometric_shapes #227: ROS 1 releases

peci1, 2023-01-10:
Both Melodic and Noetic haven't seen a release for over a year. There were a few new things merged in in the meantime. @rhaschke can I ask you to do a release? Especially Melodic's time is getting short before the EOL...

peci1, 2023-03-31:
@rhaschke @tylerjw tick tock, one more Melodic Sync has slipped through... Only 2 remaining. Could you please have a look at preparing new ROS 1 releases?

rhaschke, 2023-03-31:
I will do a release now. Sorry for the delay. I guess the current melodic-devel branch is stable on your side, isn't it?

peci1, 2023-03-31:
Thanks! Yes, Melodic has everything merged in and should work.

A quick local test says:

    Summary: 194 tests, 0 errors, 0 failures, 0 skipped

closed 2023-03-31 by rhaschke