moveit/geometric_shapes #243: Jazzy regression in RHEL9 due to missing `fcl-devel`

marcoag, 2024-06-18:
`fcl-devel` is not available in RHEL9: https://pkgs.org/search/?q=fcl-devel

So I just wanted to give a heads up that this is causing a regression for `geometric_shapes` on its version `2.2.0-1` on `RHEL9` [released recently for Jazzy](https://github.com/ros/rosdistro/pull/41569): https://build.ros2.org/view/Jbin_rhel_el964/job/Jbin_rhel_el964__geometric_shapes__rhel_9_x86_64__binary/11/console

Since the build was triggered by a version bump nothing has invalidated the previous version yet, so the previous package should be available in `RHEL9`. However this still breaks packages downstream like [robot_calibration](https://build.ros2.org/view/Jbin_rhel_el964/job/Jbin_rhel_el964__robot_calibration__rhel_9_x86_64__binary/29/console) and they will be invalidated.

Ideally we should fix the octomap problem. We are looking also at the options of having `fcl-devel` released for `RHEL 9`.

Since reverting back to an earlier version seems to break the package for `Ubuntu Noble` and the fixes might take sometime, I went ahead with a [Jazzy sync](https://discourse.ros.org/t/new-packages-for-jazzy-jalisco-2024-06-18/38240) despite this issue.

peci1, 2024-06-18:
Hmm, this is bad. I guess it also affects #239 .

marcoag, 2024-06-24:
Yup, that one would be affected as a downstream package.

marcoag, 2024-06-26:
Heads up that this package is currently in [regresison on RHEL for jazzy](http://repo.ros2.org/status_page/ros_jazzy_rhel.html?q=REGRESSION) and will be removed from the repos in the next sync (coming soon), since the issue should be solved by the latest change to use the system version of octomap (https://github.com/moveit/geometric_shapes/pull/244), I would suggest a release of this as soon as possible to avoid it being removed.

rhaschke, 2024-06-26:
Release triggered: https://github.com/ros/rosdistro/pull/41817

closed 2024-06-26 by rhaschke

marcoag, 2024-06-26:
@rhaschke also broken for `jazzy`, so it would be good to release there.

rhaschke, 2024-06-26:
https://github.com/ros/rosdistro/pull/41820

peci1, 2024-11-16:
Am I right that RHEL 9 now has fcl-devel ? So this would no longer be a problem? https://rhel.pkgs.org/9/epel-x86_64/fcl-devel-0.7.0-6.el9.x86_64.rpm.html