moveit/geometric_shapes #251: Latest Noetic forward ports to ROS 2 affected MoveIt 2 CI

sea-bass, 2024-11-26:
The forward ports in  #239 have caused a regression in MoveIt 2 CI, which uses `geometric_shapes` for path constraints.

Interestingly enough, the only breakage has to do with running the `main` branch of MoveIt 2 on ROS 2 Humble. I suspect there is some dependency back in Ubuntu 22.04 that makes things not happy.

See https://github.com/moveit/moveit2/issues/3130 for more details.

I reverted the entire commit and it does solve the issue. I'm now running a bisection of the changes to pinpoint exactly what it was. Will update this issue.

---

sea-bass, 2024-11-26:
Yep. It's this: https://github.com/moveit/geometric_shapes/commit/b00b1e5

@peci1 if you don't mind, I'm going to cherry-pick this to `ros2` now.

peci1, 2024-11-26:
Thanks for the quick action.

I actually asked if breaking ABI on the ros2 branch is a problem, but none of the maintainers objected: https://github.com/moveit/geometric_shapes/pull/239#pullrequestreview-2013504934 .

Now the question is when will be the right time to remove the ABI compatibility fix... When a new ROS2 distro is branched off?

sea-bass, 2024-11-26:
Given that the failure was only on Humble, I suspect it will be OK as soon as MoveIt's `main` branch is deemed no longer compatible (it still is!).

closed 2024-11-26 by sea-bass