moveit/geometric_shapes #255: Likely regression with dependencies in 2.3.0

mikeferguson, 2024-12-04:
#239 adds a `<build_depend>` on `libfcl-dev` and `<exec_depend>` on `libfcl` - however when 2.3.0 was released, `robot_calibration` debians stopped building for Jazzy and Rolling with the following error:

```
/usr/bin/ld: cannot find -lfcl: No such file or directory
```

I don't directly do anything with fcl, but do find_package on geometric_shapes and link against (to utilize the mesh loader).

`geometric_shapes` has only three other (released) downstream users: `moveit_core`, `moveit_ros_occupancy_map_monitor`, `moveit_ros_visualization` - the latter two of which depend on `moveit_core`. `moveit_core` has a `<depend>` on `libfcl-dev`. Thus, these continued to build properly.

I've temporarily patched my build by adding a `<depend>libfcl-dev</depend>` to the package.xml of robot_calibration - but not sure if this maybe indicates a deeper problem that could catch other users of `geometric_shapes` off guard.

peci1, 2024-12-04:
I'm not sure why robot_calibration pulls fcl into linker. For geometric_shapes, it is (or should be) a private, non-transitive dependency. So it seems like something else pulls it in without declaring it?

peci1, 2024-12-04:
Hmm, maybe this:

https://github.com/moveit/geometric_shapes/blob/ros2/CMakeLists.txt#L93

should contain PRIVATE?

peci1, 2024-12-05:
Indeed, in http://packages.ros.org/ros2-testing/ubuntu/pool/main/r/ros-jazzy-geometric-shapes/ros-jazzy-geometric-shapes_2.3.0-1noble.20241203.001623_amd64.deb , in file `data.tar.zst/opt/ros/jazzy/share/geometric_shapes/cmake/export_geometric_shapesExport.cmake`, I see:

```
set_target_properties(geometric_shapes::geometric_shapes PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include/geometric_shapes;/opt/ros/jazzy/include;/usr/include/eigen3;/usr/include;/usr/include/libqhull_r;${_IMPORT_PREFIX}/include"
  INTERFACE_LINK_LIBRARIES "fcl;console_bridge::console_bridge;geometry_msgs::geometry_msgs__rosidl_generator_c;..."
)
```

So `fcl` gets exported as interface link library.

I'll try to submit the fix I've proposed earlier.

peci1, 2024-12-05:
See if #256 fixes the issue.

mikeferguson, 2024-12-05:
@peci1 I don't have a great way to test that PR - since merely having libfcl-dev installed fixes the issue - and I don't think I want to tear up that much of my system to test it...

mikeferguson, 2024-12-06:
Appears that 2.3.1 (which is now built on testing) does fix the issue - https://github.com/mikeferguson/robot_calibration/pull/188

closed 2024-12-06 by mikeferguson

peci1, 2024-12-06:
Great, thanks for testing!

PR #256 (2024-12-06): Do not export fcl dependency
Hopefully closes #255.

It seems we have erroneousely exported `fcl` as interface link library when it should only be private (it is only used in `obb.cpp` and in no header files).